kkbin505/Arduino-Transmitter-for-ELRS #6: About RC Channel

mrAyaz, 2022-07-08:
thanks for this wonderful project. Can you tell me how many channels this project supports. I need 10 channels is this possible?

kkbin505, 2022-07-12:
Thanks for your support.
The Arduino nano I used have (A0 to A7) 8 analogy IO and (D2 to D13) 12 digital IO, so it can have 20 channels in theory, but crsf current only transfer 16 channels. 
So my answer for this question for this question is yes, it can support 10 channels, you can add more channels use free Its and add code your self.

mrAyaz, 2022-07-12:
Can you show? You have a example?

kkbin505, 2022-07-16:
You can add more channels by adding more io in the config:
const int DIGITAL_PIN_SWITCH_AUX4 = 5;(pin 5 is a free io pin)

add this line in main to avoid float
pinMode(DIGITAL_PIN_SWITCH_AUX4, INPUT_PULLUP);

Add :
AUX4_value = digitalRead(DIGITAL_PIN_SWITCH_AUX4);
and:
rcChannels[AUX4] = (AUX4_value == 0) ? CRSF_DIGITAL_CHANNEL_MIN : CRSF_DIGITAL_CHANNEL_MAX;
 
you can add upto 12 aux channels

kkbin505, 2022-08-20:
The limilation is only the IO pins, close the issue

closed 2022-08-20 by kkbin505