kkbin505/Arduino-Transmitter-for-ELRS #8: About ExpressLRS 3.0

mrAyaz, 2022-09-25:
Hi dear Developer. please tell me SimpleTX support Express LRS 3.0 or not. And now i make a RC baset on SimpleTX but my TX module (Happymodel 2.4 GHz) blink only Blue LED. why? please help. i connect only joystic Gimbal 4 chanel.

kkbin505, 2022-09-27:
SimpleTX support ELRS 3.0 in the way that gimbal data can be transfer to the TX moduel, because the communication between TX and the transmitter dosen't change since ELRS 1.0 (CRSF protocol). But the simpleTX do not support power or fresh rate config for 3.0 yet (plan to do that). 
For the Happymodel moduel,  do you follow every step in my readme file, I can not test because I use DIY module (JR, nano and super slim) all of them works fine, most importand thing is uncheck - uart invert, because simpleTX only support un inverted uart on avr micro controller I use.

tmb-ix, 2023-06-15:
Hello.
on version 3.2, the modules establish a connection. But the channels are not transmitted. Tell me, is there a solution?

galaxyrepairman, 2023-11-23:
Hello, according to your video operation, it seems that there is no need to burn the PPM to CRFS protocol into Arduino. Is it actually necessary to burn it into Arduino?

galaxyrepairman, 2023-11-23:
我使用elrs2.5.1版本无法对频

sercona, 2024-08-02:
> Hello. on version 3.2, the modules establish a connection. But the channels are not transmitted. Tell me, is there a solution?

is it possible that you are using the wrong pad on the elrs 'rx as tx' chip?

this happened to me.  I wired the tx uart pad on the nano cpu to the tx pad on the elrs module and that was incorrect.  I tried using the rx pad on the elrs module and finally I was able to see data come thru.  so that's one thing to check.  only 1 wire for data is used on the elrs module.  its clear that the tx (data-out) is the right one for nano since the nano is in 'transmit only' mode.  but it was not clear to me which line on the elrs module was the one to use.  try one, then try the other.  that allowed me to see data come thru.

kkbin505, 2024-11-22:
Elrs 3.0 and 2.0 use the same rc data protocal called crsf ground protocal. So currently it should work on elrs 3.0 in theory. It is only the guesture configuration will not work on elrs 3.0. I am working on update it .

kkbinmail, 2025-02-12:
ELRS 3.0 support added in the main branch