kkbin505/Arduino-Transmitter-for-ELRS #10: Custom RC with switches only

not7cd, 2023-06-14:
Hey, I was building a custom case with only switches and used your code as a library. I have some issues sending the actual commands and setting a proper Packet Rate. 

Here are the specs of our project:
* Arduino MEGA, the source code shares process time with ethernet connection and MQTT messages.
* We are using Nano module, I had no luck with JR micro module
* ExpressLRS 3.2.1. Which version did you use?
* I have set different Packet Rate

What is happening:
* Handset detected by Nano TX Module
* Connects to RX module
* No inputs visible on the flight controller side

Maybe you have some ways to debug this, as 
Our source code: https://gitlab.com/SimLE/simba/control-panel-software

I would be grateful if you could help soon.

MattiasKallio, 2023-06-14:
Hi!

I'm not sure why I got this message, I am not involved in that project at
all.

But it's weirdly close to things I do, so I had to go through some code to
be sure. 😂

Cheers
//Mattias

kkbin505, 2024-11-12:
Did you make it?

ne9n, 2024-11-12:
no not yet.  It is on deck for a winter project.
I have several never used old RC transmitters I would like to convert to a
new encoder and RF deck.  I wanted to use 50 Mhz as I am a ham but saw ELRS
and am looking at it.

I am an EE manager and my day job is very close to this so, it will be a
fun project.

closed 2024-11-22 by kkbin505