kkbin505/Arduino-Transmitter-for-ELRS #13: Support for Un-Inverted CRSF

inx199, 2023-12-03:
Iam using custom build firmware on my R9M, and i cant change the UART_INVERTED option, I know arduino can't output Un-inverted CRSF, but can it be done via hardware that can simply inverse the signal? or by using ESP32 which support signal inversion?

sercona, 2024-08-02:
yes, the simplest solution is to use a logic gate and literally invert the uart signal.  note you may need 3.3v logic (likely so).

kkbin505, 2024-08-30:
> Iam using custom build firmware on my R9M, and i cant change the UART_INVERTED option, I know arduino can't output Un-inverted CRSF, but can it be done via hardware that can simply inverse the signal? or by using ESP32 which support signal inversion?

For arduino based simple TX, you need a logical invert gate to do that.
If you need to implement uart-inverted in the TX side, my suggestion is to use ESP32 based simple TX instead which have build inverted uart option.
https://github.com/danxdz/simpleTx_esp32

closed 2024-08-30 by kkbin505