kkbin505/Arduino-Transmitter-for-ELRS #15: Analog inputs not working

ArshaanTech, 2024-07-25:


sercona, 2024-08-02:
you must calibrate, first.  I think that was something I did wrong, also.

calibration tells the system the min and max of the sweep.  if it defaults to all the same midpoint, no sticks will be seen as 'moving'.

I think that's the issue.

ArshaanTech, 2024-08-03:
I tried to caliberate it aswell, 
I uploaded the code after with caliberation, but it doesn't seem to work :(

ArshaanTech, 2024-08-03:
even during caliberation, it shows no changes.

sercona, 2024-08-03:
did you connect the a-ref pin on the arduino nano to 3.3v (right next to it)?

are you sure your gimbals are not wired backwards?  I have radiomaster boxer gimbals and red is MINUS and black is PLUS.  dammit.  stupid radiomaster.  if your hall gimbals (which care about polarity) are not wired right you wont get any analog variation out.

simplest test, just run a short test program that reads from a single analog in and prints its output.  just that.  if you cant get a single analog in with the joystick you need to figure out if its voltage or what the issue is.

you could start with a cheap pot (1k or 10k) and try connecting its wiper to the analog input pin and when you connect ground to one end and 3.3v to the other, you should see values change.

ArshaanTech, 2024-08-03:
No i did not connect the a-rf pin on arduino nano to 3.3v, Is it important?
My gimbals are working fine with a different code for a transmitter.
I will try connecting a-ref to 3.3v.

sercona, 2024-08-03:
the code wants to use an external voltage ref.  so yeah, you HAVE to since it assumes a 3.3v ref; the same supply that you use for the gimbal positive rail.

sercona, 2024-08-03:
if your code has:

  **analogReference(EXTERNAL);**

then yes, you need those 2 pins shorted (3.3v and a-ref)

ArshaanTech, 2024-08-04:
Thank you so much, I just got it working!

closed 2024-08-04 by ArshaanTech