kkbin505/Arduino-Transmitter-for-ELRS #17: Not able to bind.

ArshaanTech, 2024-08-18:
Hi, 
I am not able to bind my receiver with my my DIY Tx module and simple TX. How do i put my Tx into bind mode??

sercona, 2024-08-18:
elrs v2 and v3 seem to have changed and I dont think the stick based binding works anymore.  what I've done is to go into the web ui for the rx-as-tx module and define what the boot button does.  if you dont have a button, you should have a pad and ground to short.  wire up a button to that and mount it on the tx chassis.

triple power cycle the rx and it should be waiting in bind mode.  press the bind button on your tx briefly and the bind process should complete.

note, 250hz is the default for the rx-as-tx and so your rx will have to 'settle' on 250hz to get quicker bind.  what often is helpful is to use your main tx (edgetx) and set its elrs to 250hz and get your quad to bind.  once it binds at 250, go into elrs lua menu on the tx and change it from 250 down to something and then back to 250.  all while the quad is still on and had a bind lock from the edgetx tx.  this will cause short failsafes and once you are bound back to 250, you can power off your edgetx radio and then power on the diy one.  the bind should happen faster now (the quad remembers the 250 and your tx only will talk at 250 unless the host protocol is updated to v3)

ArshaanTech, 2024-08-18:
Can you tell me how can i go to the webui rx as tx and define the bind button, I am really new and i don't know how to do that...
I don't have any ratdio with edgeTX. I am currently making this SimpleTX...
Thank you.

sercona, 2024-08-18:
it will be more helpful to have a 'master' edgetx radio just to make sure your rx is working ok.  if you have only your diy tx, when things dont work, how do you know which end is the problem?

the rx should go into wifi mode after 30 or 60 seconds.  it will blink fast.  go to your phone(etc) and look for the expresslrs wifi AP.  connect to that, then either go into its web ui (10.0.0.1) or give it your own wifi ssid/pass and it will reboot into your network.  either way, the web ui will have a tab that talks about the button that is part of all elrs rx's.  even if just a solder pad marked 'boot'.

ArshaanTech, 2024-08-18:
> it will be more helpful to have a 'master' edgetx radio just to make sure your rx is working ok. if you have only your diy tx, when things dont work, how do you know which end is the problem?
>
> the [...]

Okay i did that... What should i do next how do i bind my Tx to that Rx?

sercona, 2024-08-19:
for the basic question, you might want to ask on the elrs discord.  read the faq for elrs and you should get all you need.

kkbin505, 2024-08-30:
> Hi, I am not able to bind my receiver with my my DIY Tx module and simple TX. How do i put my Tx into bind mode??

Simple TX does not support TX bind and have no plan to support in the future. How I bind the TX to the RX is ulitize the ELRS binding phase. All my receives are set the same bind pahse as the TX, they will automantically bind when powered up.

kkbin505, 2024-08-30:
> Okay i did that... What should i do next how do i bind my Tx to that Rx?

The ELRS repo or discord is a better place to discusse this topic. There are more people can help you there.

closed 2024-08-30 by kkbin505