kkbin505/Arduino-Transmitter-for-ELRS #20: connect elers2.4 moudel

naorwaiss, 2024-11-16:
hi i try to fllow your project - and connect my expresslrs modole to a arduino and i stack with no handset at the moudel -  you know this problem ?

kkbin505, 2024-11-22:
Which elrs version did you use, and what hardware did you use.

kkbin505, 2024-12-13:
Please first flash my repo without any modification to see if your module is working to make sure this issue is not related to the moudle. If it is working, you can add your own code.

kkbin505, 2025-03-12:
> hi i try to fllow your project - and connect my expresslrs modole to a arduino and i stack with no handset at the moudel - you know this problem ?

Did you sovled this issue?

kkbin505, 2025-07-29:
It is not a code issue. It is hardware issue.

closed 2025-07-29 by kkbin505