kkbin505/Arduino-Transmitter-for-ELRS #23: Port SimpleTX From Arduino Nano to ESP32 or STM32 for more features and to support Full-Duplex for Telemetry for ELRS Using generic OLED Display.

Okami-423, 2025-01-27:
Hi, kkbin505 
i have seen your simpleTX video and Also Build my own simpleTX and i love it for how simple it is to use but i can not ignore how underpowered it is compared to the boards available in same price bracket in 2025, So i would like to see you build a version of simpleTX for ESP32 or STM32 with telemetry but i do not possess such coding skills to build it on own, So i would love it if you could build it.
Thank You

basha318i, 2025-02-10:
I'm trying to port this to also be used with RC cars, where merely throttle & steering are used, it would be nice to get a cheap RC car controller & convert it to ELRS! Thanks.

kkbin505, 2025-02-12:
I am not planing to add complex feaures to simpleTX. 
BTW other people have already implement the telementry etc. 
Please check a ESP32 based SimpleTX
https://github.com/danxdz/simpleTx_esp32
and STM32 based simpleTX
https://github.com/danxdz/simpleTx_esp32

closed 2025-02-12 by kkbin505

Okami-423, 2025-02-12:
Oh really i did not know about it.
Thanks for sharing other links
Really appreciate IT.