kkbin505/Arduino-Transmitter-for-ELRS #27: CH7

55johnchiu, 2025-04-02:
希望多一個三位掣 CH7上

kkbin505, 2025-04-05:
如果要实现三段，还需要一个pin，你希望是哪个pin

55johnchiu, 2025-04-07:
9,pin可以嗎?

55johnchiu, 2025-04-07:
我在2.x新增,正常運作

// pins that used for the switch
const int DIGITAL_PIN_SWITCH_ARM = 2;  // default arm switch for Elrs is channel 5 (digital 1)
const int DIGITAL_PIN_SWITCH_AUX2_HIGH = 7; // 3 stage switch for mode setup
const int DIGITAL_PIN_SWITCH_AUX2_LOW = 8; //
const int DIGITAL_PIN_SWITCH_AUX3_HIGH = 4; //const int DIGITAL_PIN_SWITCH_AUX3 = 4;
const int DIGITAL_PIN_SWITCH_AUX3_LOW = 9;//
const int DIGITAL_PIN_SWITCH_AUX4 = 3;  //

// pins that used for output
const int DIGITAL_PIN_LED = 5;    // in pcb v0.9 led is reused from AUX4 (remember to add 300om resistor between led and 5 digital output)
// pins that used for buzzer
const int DIGITAL_PIN_BUZZER = 6;
//----- Voltage monitoring -------------------------
// Define battery warning voltage
const float WARNING_VOLTAGE = 7.4; // 2S Lipo 3.7v per cell
const float BEEPING_VOLTAGE = 7.0; // 2S Lipo 3.5v per cell
const float ON_USB = 5.2;          // On USB power / no battery

kkbin505, 2025-04-07:
好的，这个功能我就不放到正式版本里了。

closed 2025-04-07 by kkbin505