kkbin505/Arduino-Transmitter-for-ELRS #29: Connecting a readymade module to the nano

Infilux, 2025-07-16:
Hi, I am connecting a readily available Expresslrs module to the DIY radio but it not working. I am using radiomaster nomad module. I have connected everything on the breadboad, only change is, instead of the LoRa module I am using the radiomasted nomad but its not working the way it should. Kindly let us know where is the issue. 

Product link: https://radiomasterrc.com/collections/nomad/products/nomad-dual-1-watt-gemini-xrossband-expresslrs-module

kkbin505, 2025-07-17:
Which elrs version did your tx use

Infilux, 2025-07-17:
lastest 3.5.6

Infilux, 2025-07-17:
See, I will be honest, I havent bought all the recommended hardware yet. I have got a nano on my breadbord, where I am trying to sent 1 single input from my nomad module to the RX connecte to FC(Monitering on betaflight). I just want to see that when I press a button it is reflected on the betaflight channel monitor. Once this is confirm I will place order for all the required stuff and build the complete TX. I have tried but no input are transmitted yet. The module is bound to the RX and the TX pin is connected to the s-port pin of the module and a switch is connected. Kindly guide me.